mrrwa/LocoNet #4: __vector_11 Error with Servo.h Library

pturvill, 2016-06-05:
The __vector_11 conflict between LocoNet.h and Servo.h has been discussed numerous times, but I have yet to find a solution. The marriage between these two libraries would seem to be a natural. Is there a solution anywhere In sight?

kiwi64ajs, 2016-06-06:
Yeah that would be nice to be able to use the core Servo library but as the LocoNet decoding logic currently requires the use of the ICP peripheral, and that is _only_ available onTimer1 AND the Server library ALSO uses Timer1 - that issues is not going to be resolved anytime soon.

I have been thinking of trying to changing to code to use the UART and Pin Change interrupts to sense LocoNet Rx Activity but that is a while off and as the UART on the Mega328 is already used by the bootloader it's not really viable. Perhaps for the 32u4 or other chips like the Teensy 3.x. SO for now you're stuck with the Software Servo libraries which probably don't perform as well.

You might want to consider driving the servos from an external PWM chips like these:
http://www.ebay.com/itm/16-Channel-12-bit-PWM-Servo-Drive-shield-module-I2C-PCA9685-For-Arduino-NEW-/401090900127

HTH

Alex Shepherd

closed 2016-06-06 by kiwi64ajs

pturvill, 2016-06-08:
Problem solved. A detailed Google search turned up the ServoTimer2 library by Michael Margolis (updated by Nick Bontrager) which as the name suggests, uses Timer2, and is compatible with the LocoNet library. It treats servo control a bit differently than the core Servo library, using pulse width rather than degrees to set the position, but with a tiny bit of tweaking it is now working perfectly on my homebrew Nano-based Quad_Servo_LocoNet_Driver sketch.

Thanks for the reply.

From: Alex Shepherd 
Sent: Sunday, June 5, 2016 11:53 PM
To: mrrwa/LocoNet 
Cc: pturvill ; Author 
Subject: Re: [mrrwa/LocoNet] __vector_11 Error with Servo.h Library (#4)

Yeah that would be nice to be able to use the core Servo library but as the LocoNet decoding logic currently requires the use of the ICP peripheral, and that is only available onTimer1 AND the Server library ALSO uses Timer1 - that issues is not going to be resolved anytime soon.

I have been thinking of trying to changing to code to use the UART and Pin Change interrupts to sense LocoNet Rx Activity but that is a while off and as the UART on the Mega328 is already used by the bootloader it's not really viable. Perhaps for the 32u4 or other chips like the Teensy 3.x. SO for now you're stuck with the Software Servo libraries which probably don't perform as well.

You might want to consider driving the servos from an external PWM chips like these:
http://www.ebay.com/itm/16-Channel-12-bit-PWM-Servo-Drive-shield-module-I2C-PCA9685-For-Arduino-NEW-/401090900127

HTH

Alex Shepherd

kiwi64ajs, 2016-06-08:
Excellent!

That’s a good outcome. I’ll have to take a look at that library as I’ve had to avoid the core one as well…

Alex