mrrwa/LocoNet #6: Compatibility Arduino Micro

nepix32, 2017-10-02:
Tried the Loconet-Shield from Dirkjan Kapers together with an Uno, which worked really nice.

However, with an Ardunio Micro it did not work. Then I noticed that only the Uno and the Atmega are mentioned in the ``README.md`` and in ``ln_config.h``. Would it be feasible just to adapt the firmware or is the 32u4 fundamentally different?

Realized that the Arduino Nano should be what I want (pin-headers, more compact).

kiwi64ajs, 2017-10-03:
Hi,

> On 3/10/2017, at 9:00 AM, nepix32 <notifications@github.com> wrote:
> 
> Tried the Loconet-Shield from Dirkjan Kapers together with an Uno, which worked really nice.
> 
> However, with an Ardunio Micro it did not work. Then I noticed that only the Uno and the Atmega are mentioned in the README.md and in ln_config.h. Would it be feasible just to adapt the firmware or is the 32u4 fundamentally different?
> 
> Realized that the Arduino Nano should be what I want (pin-headers, more compact).
> 
I’ve recently used one of John Plocher’s LocoShield boards with an AVR 32U4 based board (Arduino Pro Micro) and it seemed to work fine. The board I used is like this one:

	https://learn.sparkfun.com/tutorials/pro-micro--fio-v3-hookup-guide <https://learn.sparkfun.com/tutorials/pro-micro--fio-v3-hookup-guide>

I have since added an another example sketch (in version 1.0.5) with specific Serial USB Buffer handling to try and make sure I don’t drop packets etc so hopefully it’s all good. 
The modified sketch is here: 

	https://github.com/mrrwa/LocoNet/blob/master/examples/LocoLinxProMini/LocoLinxProMini.ino <https://github.com/mrrwa/LocoNet/blob/master/examples/LocoLinxProMini/LocoLinxProMini.ino>

John spun another variant of the LocoShield board specifically for the Arduino Pro Micro PCB here:

	http://www.spcoast.com/wiki/index.php/Core-Locobuffer <http://www.spcoast.com/wiki/index.php/Core-Locobuffer>     

I’ve got some blank PCBs already and just waiting for the SMD components to arrive so I can test it, but it should be any different to what I’ve had working already. 

HTH

Alex Shepherd

dirkjankrijnders, 2017-10-03:
Hi,

Did you by any chance mean my (Dirkjan Krijnders) design. I do indeed use them with a arduino micro, but there are a few modifications needed as the ICP moved with respect to the pro mini.
Let me know if you need help,

Dirkjan (Krijnders)

nepix32, 2017-10-03:
Thanks Alex,

tried the [LocoLinxProMini](https://github.com/mrrwa/LocoNet/blob/master/examples/LocoLinxProMini/LocoLinxProMini.ino) project. First it did not work because I was thrown off the trail by your reference to John Plochers LoconetShield. It seemed to suggest that I had to connect the FremoLNShield to Arduino Micro on D4 and D7. This lead to a setup which can read from loconet but not write to it.

The comment was confusing me even more: Why is the comment talking about setting RX at Pin 4 but the codeline is talking about TX on Pin 5 !?!

```
// Rx Pin on the LeoStick is D4
#define  TX_PIN 5
```

Then I just tried the following, which lead to a working LocoBuffer powered by the Arduino Micro!

For posterity:

|  Ardunio Micro | LNShield |
| --- | --- |
| GND/⏚ | GND |
| 5V | +5V |
| 4 | 8 |
| 5 | 6 |

PS: Be advised that I used [Dirkjan Kapers FremoLNShield](http://nh-finescale.nl/fremo/dcc/fremo-ln-shield/FremoLNShield.html) (sorry other Dirkjan!) for my tests.

kiwi64ajs, 2017-10-06:
Hmmm... I'll try and clean that up, too much copy/paste or rename and modify...
Maybe it might be better to call it LocoLinx32U4 as the pro/mini/micro terms are used/applied a little liberally and its become ambiguous now.

kiwi64ajs, 2017-10-07:
I've renamed the LocoLinxProMini example to LocoLinx32U4 and added a lot more comments and some conditional compilation options to make using LEDs optional.
It's tagged as version 1.0.6 so it should appear in the Arduino Library manager soon.
I've also removed the original LocoLinxLeo (Leonardo/LeoStick) as that code was not as good and is really superseded by the LocoLinx32U4 example.
Please take a look and let me know if it's an overall improvement

nepix32, 2017-10-09:
Thanks, 

Tried the [LocoLinx32U4](https://github.com/mrrwa/LocoNet/blob/master/examples/LocoLinx32U4/LocoLinx32U4.ino) project from 1.0.6 release on Dirjan Kapers FremoLNShield with an Arduino Micro.

|  Arduino Micro | LNShield |
| --- | --- |
| GND/⏚ | GND |
| 5V | +5V |
| 4 | 8 |
| 5 | 6 |

Be advised that the [Shield by John Plocher, 2.0](http://www.spcoast.com/wiki/index.php/File:Eagle-Core-Locobuffer-2.0-Core-Locobuffer.sch.x.png) does not match the pinout as used in the sketch!

| Function |  Sketch 1.0.6 | Johns Shield |
| --- | --- | --- |
| LNRX | D4 | D4 |
| LNTX | D5 | D7 |  

Again thanks for the work!

closed 2017-10-09 by nepix32

kiwi64ajs, 2017-10-09:
> Be advised that the Shield by John Plocher, 2.0 <http://www.spcoast.com/wiki/index.php/File:Eagle-Core-Locobuffer-2.0-Core-Locobuffer.sch.x.png> does not match the pinout as used in the sketch!
> 
> Function	Sketch 1.0.6	Johns Shield
> LNRX	D4	D4
> LNTX	D5	D7

Ah yes - you’re right.

On my breadboard prototype of John's new Core-Locobuffer 2.0 I had used D4 & D5 and I had assumed that persisted but I guess that is consistent with his UNO Shield  but looking again at the final layout I see Tx is on D7 not D5 as you rightly point out.

I’ll change it again so it’s right for that shield… ;)

Alex