mrrwa/LocoNet #7: Anyone familiar with the ESP8266 and ESP32?

Merdeka, 2017-11-27:
Anyone familiar with the ESP8266 and ESP32?

I can't figure out how to port the ln_sw_uart to the ESP because I'm not skilled enough in the ICP Interrupt and timer part of the Arduino and ESP.

I have been looking at:
https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/include/esp32/esp_intr_alloc.h
https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/include/freertos/freertos/timers.h
But have no idea to be honest.

Are there others that have plans to port the LocoNet library to the ESP8266 or ESP32?

Cheers,

Timo

Merdeka, 2017-11-27:
I did manage to get LocoNet working on the ESP32 using WiFi udp. But I would really like to use a 'real' LocoNet interface to connect to other LocoNet hardware using standard LocoNet wiring.

Also got DCC++ to compile on the ESP32. My Goal is to build a DCC++ command station with LocoNet.

kiwi64ajs, 2017-11-28:
Hi Timo,

> On 28/11/2017, at 11:57 AM, Merdeka <notifications@github.com> wrote:
> 
> I did manage to get LocoNet working on the ESP32 using WiFi udp. But I would really like to use a 'real' LocoNet interface to connect to other LocoNet hardware using standard LocoNet wiring.
> 
I’ve spent some time studying the ESP8266 and the ESP32 hardware UART capabilities in terms of being able to directly support LocoNet and I’d written off using the ESP8266 as it only has the one core and has times when its busy with the Wi-Fi, so could be a bit non-deterministic, which is BAD for bit-bashed comms. Also the ESP8266 doesn’t have some of the features the ESP32 has which should be helpful for LocoNet.

The ESP32 hardware UART should be able to be used for BOTH LocoNet Rx and Tx as it has several new (compared to the ESP8266) features that should allow us to do the required collision sensing and also invert the Tx output logic, so we can use a simple Open Collector output driver for the Tx line.

I had started to refactor the LocoNet library as a new project here:
	https://github.com/mrrwa/LocoNet2 <https://github.com/mrrwa/LocoNet2>

It’s not completed yet and I’ve done subsequent changes in the current LocoNet library that are now missing so I need to resync the changes and resume working on the LocoNet2 library. 

The idea was to move all the core non-platform-specific logic into base classes and allow the platform specific logic to be defined as either separate derived classes or perhaps a separate LocoNet Interface class that is registered with the LocoNet core classes.
> Also got DCC++ to compile on the ESP32. My Goal is to build a DCC++ command station with LocoNet.
> 
Cool.

I’ve actually got some code for the ESP8266 that supports the WiThrottle TCP/IP protocol and I was working towards making a Wi-Fi to LocoNet interface, but got stuck on the LocoNet interfacing using the 8266 UART. So then I figured I’d just wait until the ESP32 arrived and was supported well in the Arduino environment. That is happening now, so it’s probably time to resume the work…

HTH

Alex

Merdeka, 2017-11-28:
> The ESP32 hardware UART should be able to be used for BOTH LocoNet Rx and Tx as it has several new (compared to the ESP8266) features that should allow us to do the required collision sensing and also invert the Tx output logic, so we can use a simple Open Collector output driver for the Tx line.

That would be great. I was first thinking of using an Arduino as a sort of LocoNet co-processor for the ESP but cancelled that idea because it's just not a good solution for small hardware design.
In my testing with LocoNet UDP over WiFi I also went straight to the ESP32 because of the dual core design. Couldn't get it to work reliably in the ESP8266.

> The idea was to move all the core non-platform-specific logic into base classes and allow the platform specific logic to be defined as either separate derived classes or perhaps a separate LocoNet Interface class that is registered with the LocoNet core classes.

That would be the best solution in the long run. Because we started with the AVR then ARM Cortex came around, now it's the Xtensa LX6. Who knows what will be the next big thing in microcontrollers. Platform specific logic in separate derived classes is the way to go to future proof the Library

Cheers,

Timo

mcmadhatter, 2018-01-20:
Hi Timo
Not sure if you are subscribed to the Loconet Hackers yahoo group, but just in case you are not:

https://github.com/mcmadhatter/LocoNet2/tree/development

Jeffery

mikaelsundin, 2018-06-30:
I think it may be possible to use ESP8266 hardware UART to run Loconet collision on byte level, some tricks may needed to get everything to work. Interrupt must be used to handle transmission and receiving + maybe a interrupt pin on RX for bus idle also. 

Receiving still needs a real comparator outside, type LM393 + some resistors.
The ESP8266 hardware is capable to invert TX (bit UCTXI in U0C0).

If i get the time to test something with ESP8266 i will feedback again.

kiwi64ajs, 2019-02-01:
This repo https://github.com/mcmadhatter/LocoNet2/tree/development is probably the way to go on the ESP32 for now

closed 2019-02-01 by kiwi64ajs

positron96, 2020-05-02:
Hi, I'm interested in this as well. Right now I have a Arduino nano + esp8266 + nrf24 setup for my command station. Nrf24 handles custom wireless protocol for stationary decoders, Arduino runs modified dcc++ code,  Esp8266 handles WiThrottle protocol and dcc++ tcp/up proxying. 
I intend to replace nrf24 with loconet and replace esp8266+Arduino with esp32.