mrrwa/LocoNet #21: Version 1.1.6 doesn#t compile for Atmega328, wants LN_RX_PIN_NAME

fhennies, 2021-03-19:
Hi,

I tried to compile a simple program using 1.1.6, and get the following error. With 1.1.2 it compiles just fine. It seems that something went wrong in the board definitions in ln_config.h or ln_sw_uart.cpp

`lib\LocoNet\utility\ln_sw_uart.cpp: In function 'void initLocoNetHardware(LnBuf*)':
lib\LocoNet\utility\ln_sw_uart.cpp:397:10: error: 'LN_RX_PIN_NAME' was not declared in this scope
  pinMode(LN_RX_PIN_NAME, INPUT);`

bjoernrennfanz, 2021-03-21:
Hi,

I think there was a merge error here, I looked at the code, the error is in line 393 - 399. Please replace with the following code in ln_sw_uart.cpp. I will create an Pull-Request to fix the error...

`
#if defined(ESP8266)
	pinMode(LN_RX_PORT, INPUT);
#elif defined(STM32F1)
	pinMode(LN_RX_PIN_NAME, INPUT);
#else
  	cbi( LN_RX_DDR, LN_RX_BIT );
#endif
`

I hope this helps you.

fhennies, 2021-03-21:
Thanks, the patch works for me.

closed 2021-03-21 by fhennies