mrrwa/LocoNet #25: No RX until at least on TX

Digital-MoBa, 2021-09-30:
Since the newer versions there is a problem if you only use a device that show the RX LocoNet data. 
To test this I used without any change the example "LocoNetMonitor" and upload it. It shows just "LocoNet Monitor" nothing else, no RX data is reported on the serial monitor. 
If you just add one LocoNet TX sending like "LocoNet.reportSwitch(3);". You can do this inside the setup() function, all data send from this point on will shown on the serial monitor.

devel-bobm, 2021-09-30:
Does your sketch work without modification with older versions of the library?

Can you tell whether your device is trying to drive a 0 bit to LocoNet with the latest library?  If so, you may need to force the pin controlling the LocoNet TX transistor to its opposite state in the beginning of the sketch.

Digital-MoBa, 2021-10-01:
I just did a short test with the old version 1.0.7. There the receive works directly, without a need of sending an TX packet:
_LocoNet Monitor
RX: A0 07 1D 45 
RX: A0 07 22 7A 
RX: A0 07 26 7E 
RX: B0 02 00 4D Switch Request: 3:Thrown - Off
RX: A0 07 25 7D 
RX: A0 07 21 79 
RX: A0 07 1C 44_ 
I use a normal LocoNet interface that drive a 1 bit to the LocoNet library. All test I did with an Arduino UNO and MEGA.

steamboating, 2021-10-11:
I have exact the same issue here
- Arduino 1.8.16 and Visual Studio PlatfomIO
- Arduino NANO old bootloader
out of the LOCONET-box example  "LocoNetMonitor"
- works with Loconet 1.1.4 
- does not work with Loconet 1.1.6/1.1.7 (different numbers for same release in library.json and library.properties)

Adding the line
     'LocoNet.reportSwitch(3);' 
in the Arduino Setup works for me also!

By the way, 1.1.4 and 1.1.6/7 does not work with a 
 LocoNet.init(9);
only with 
 LocoNet.init();