mrrwa/LocoNet #26: ESP8266 no packet receive

Digital-MoBa, 2021-11-08:
I try to use the ESP8266 Wemos D1 Mini with the LocoNet library. All data that is send with the library is correctly received over PIN D6 (GPIO12).
All other packets, send by other LocoNet clients aren’t received! 
But I cloud figure out that the library with LocoNet.available() give TRUE if there is send a packet by another device. But LocoNet.receive() doesn’t report any valid LnPacket from the buffer back. 
Why the packet is in the buffer but can’t read out with the receive function?
(Notice: With a Arduino MEGA/UNO all packets are received with this library and the same source code)

kiwi64ajs, 2021-11-13:
Hi Philipp,

> On 9/11/2021, at 3:20 AM, Philipp Gahtow ***@***.***> wrote:
> I try to use the ESP8266 Wemos D1 Mini with the LocoNet library. All data that is send with the library is correctly received over PIN D6 (GPIO12).
> All other packets, send by other LocoNet clients aren’t received!
> But I cloud figure out that the library with LocoNet.available() give TRUE if there is send a packet by another device. But LocoNet.receive() doesn’t report any valid LnPacket from the buffer back.
> Why the packet is in the buffer but can’t read out with the receive function?
> (Notice: With a Arduino MEGA/UNO all packets are received with this library and the same source code)
> 
1) Hmmm… I’ve not personally used the library with an ESP8266 chip, but looking at the source code the Rx pin is defined to be “D6” here:

	https://github.com/mrrwa/LocoNet/blob/master/utility/ln_config.h#L224 <https://github.com/mrrwa/LocoNet/blob/master/utility/ln_config.h#L224>

and the Default TX Pin is defined to be “D7” here:

	https://github.com/mrrwa/LocoNet/blob/master/LocoNet.cpp#L113 <https://github.com/mrrwa/LocoNet/blob/master/LocoNet.cpp#L113>

So if you have those pins wired to your LocoNet interface circuitry then you should be ok.

2) Another issue could be with the LocoNet Interface circuit are you using as you may have to invert the LocoNet electronic signal polarity compared to the logic (active High/Low) in the library.

Check the logic of the circuit first and maybe experiment with commenting/uncommenting these two lines that Hans Tanner added:

	https://github.com/mrrwa/LocoNet/blob/master/utility/ln_config.h#L87 <https://github.com/mrrwa/LocoNet/blob/master/utility/ln_config.h#L87>

	https://github.com/mrrwa/LocoNet/blob/master/utility/ln_config.h#L92 <https://github.com/mrrwa/LocoNet/blob/master/utility/ln_config.h#L92>

3) Failing that let me know the version of the library you are using as there maybe some bugs as there has been quite a bit of change recently with STM31 and ESP8266 / ESP32.

HTH

Regards

Alex Shepherd

m: +64-21-777764
e: ***@***.***

Digital-MoBa, 2021-11-16:
Hi Alex,

thank you for your replay. 
The PIN configuration works and there is data received that is send by the ESP8266. If I disconnect the RX PIN (D6) the sending doesn't work, because the library can't check the data signal.
I'm using the same LocoNet Interface that I use with the Arduino UNO and MEGA. [https://pgahtow.de/w/Loconet#/media/Datei:Arduino-LocoNet_Interface.png](url)
So the logic should be correct, because also the sending (TX data and verify with RX) from the ESP8266 works fine.
What I find strange is that if another device send data to the ESP8266, it detects them with LocoNet.available() == true. But the data itself can't read with LnPacket = LocoNet.receive();

I wrote a small test application:
[ESP_LocoNet_Test.zip](https://github.com/mrrwa/LocoNet/files/7546085/ESP_LocoNet_Test.zip)

Here is the result when it's working on a **Arduino MEGA:**
`LocoNet Monitor
Send Request: 
DATA 
RX: BB 01 00 45 
DATA 
DATA 
RX: 81 7E 
DATA 
DATA 
DATA 
DATA 
DATA 
DATA 
DATA 
DATA 
DATA 
DATA 
DATA 
RX: E7 0E 01 13 57 00 20 07 00 00 00 00 00 74 
Send Request: 
DATA 
RX: BB 01 00 45 
DATA 
DATA 
DATA 
DATA 
DATA 
DATA 
DATA 
DATA 
DATA 
DATA 
DATA 
DATA 
DATA 
DATA 
RX: E7 0E 01 13 57 00 20 07 00 00 00 00 00 74 
`

If I'm using the same sketch on the **ESP8266** Wemos then I get only this:
`Send Request: 
DATA 
RX: BB 01 00 45 
DATA 
Send Request: 
DATA 
RX: BB 01 00 45 
DATA 
Send Request: 
DATA 
RX: BB 01 00 45 
DATA 
DATA 
Send Request: 
DATA 
RX: BB 01 00 45 
DATA 
Send Request: 
DATA 
RX: BB 01 00 45 
DATA 
`
There is no answer from the master to the slot request received. 

regards Philipp

Digital-MoBa, 2022-01-12:
Hi Alex,

yesterday I got an new idea: I try to set up two ESP8266 with a LocoNet Interface and see if they can communicate with each other. Yes, they both can send messaged and also receive them without problems! Is it that simple?

I would answer this question by myself, no not at all. The major problem is a timing issue with the timer1 on the ESP8266. The timer1 is set with #define LN_BIT_PERIOD ((F_CPU / 16) / 16666) in the same way like for the other microcontroller. 
[https://github.com/mrrwa/LocoNet/blob/master/utility/ln_config.h#L95]
One issue is that the calculated timer1 speed doesn't fit to the specification. But the big fault is that on the ESP8266 you can config the CPU frequenz to 80Mhz oder 160Mhz and this has no effect to the timer speed!

With my testing i found the best solution if the timer1 is used without a divider in _ln_sw_uart.cpp_ (line 439) `timer1_enable(TIM_DIV1, TIM_EDGE, TIM_SINGLE);`
[https://github.com/mrrwa/LocoNet/blob/master/utility/ln_sw_uart.cpp#L439]

Also we need to fix with this line: `#define LN_BIT_PERIOD 4720`
[https://github.com/mrrwa/LocoNet/blob/master/utility/ln_config.h#L95]
Without the divider this value can be much more accurate. The value is tested with both cpu frequencies and DAISY II, IntelliBox and Arduino MEGA as LocoNet peripherals. 

regards Philipp

closed 2022-01-12 by Digital-MoBa