mrrwa/LocoNet #33: Could not select latest Version in Libraray Manager

pischky, 2022-05-29:
LocoNet/library.properties says that 1.1.9 is the current (latest) version.
But in the Arduino-IDE (Version 1.8.19) the latest Version I can select is 1.1.4.

Are there any release files missing? 

Yes, I know that I can download a zip and copy manually, but I assumes that it should be much easier....

[image]

kiwi64ajs, 2022-05-31:
Its pretty much just because I've merged in some pretty large and potentially disruptive PRs from others that add support for other CPUs, which I don't have setup to test at all, so I was being conservative.
However, given your prompting, I guess its probably about time they were released and if there are issues - we'll just have to sort them.
I've created the 1.19 tag and pushed that to the Git Repo, so that should propagate through tot he Arduino Library manager in the new few hours

closed 2022-05-31 by kiwi64ajs

pischky, 2022-05-31:
Thanks!
If I find something not working in 1.1.9 I will write a new Issue.

kiwi64ajs, 2022-05-31:
Cool. I just checked and the Arduino Library Manager looks to offer the 1.1.9 version for install now.

Enjoy