mrrwa/LocoNet #41: Outdated version on PlatformIO

markatk, 2023-11-28:
The lastest version in the PlatformIO registry is [1.1.6](https://registry.platformio.org/libraries/mrrwa/LocoNet). It would be nice to push the latest or newer versions to the registry as well.

kiwi64ajs, 2023-11-28:
I don’t use Platformio, so can you suggest the needed changes to achieve that so I can update the libraryRegardsAlex Shepherd

markatk, 2023-11-28:
The project already contains a [manifest file](https://docs.platformio.org/en/latest/librarymanager/creating.html#manifest). Some information may be added like the license or dependencies shown in the example `library.json`.

As I have not published anything to the PlatformIO registry myself, I can only reference the [manual about publishing](https://docs.platformio.org/en/latest/librarymanager/creating.html#publishing), but it looks like it is just the `pio pkg publish` with the correct credentials.

The manual for the manifest file and publish command is also mentioned in the README.md of this project.