mrrwa/LocoNet #44: Possible wrong size of data (packet) array?

cpm64, 2024-04-19:
Hello,

Checking the declaration of the LnMsg.data[16] . If I got it right this is the max length of a package. Searching the internet I found various answers about what is the max size of a variable length package: 16, 20. However is not clear if that size includes the OPC [0], the msg size [1] and the checksum bytes:  <OPC>,<COUNT>,<ARG1>....,<CKSUM>.

I discovered that my CS is sometimes issue an '0xe6' OPC of 21 bytes which cannot be handled by LnMsg.data[16]. 

As a suggestion, a possible enhancement could be to make it configurable via #define LN_MAX_PACKET_LEN (n) and allow the users to define their own max limit.

This is applicable to both LocoNet and LocoNet2-development branch

devel-bobm, 2024-04-19:
1) LocoNet "Message" is the correct term, not "package".

2) LocoNet message "length", for MRRwA library purposes, includes the Opcode, any data bytes, and the checksum.  This means that a valid LocoNet message can be as little as 2 bytes, or as many as 127 (or maybe 128, depending on what Digitrax says in their private LocoNet specs!) bytes.

3) A 25 year old public LocoNet "spec" has been available at the Digitrax web site.  But I did not find it this morning.  Digitrax has seemed to have been reducing the amount of useful information available on their web presence over recent years.  Perhaps they have gotten rid of it...

Be aware that the document has NOT been publicly updated since its release, and quite a large number of newer messages have been included but not publicly disclosed, including longer messages than are currently documented there.
 
4) There are known messages with greater lengths than 16.  Depending on which LocoNet devices one uses, and if one must "allow for any LocoNet message, rather than "just the ones seen on typical Digitrax hardware",  or only "old-stlye Digitrax hardware", the "longest" message _does_ change a bit.

The question about "What is the longest LocoNet message length" was recently discussed at some detail at https://groups.io/g/LocoNet-Hackers/message/11204 .  I suggest that you should refer to that LocoNet hackers thread.

(The LocoNet-Hackers group tends to be a better place to discuss LocoNet technical issues than the MRRWA group...)

5) Some "hackers" have done some things that are "not really correct", from the Digitrax perspective.  In some cases, those developments may not work "correctly" with other Digitrax hardware and/or Digitrax-licensed 3rd-party hardware/softwrae, or even other do-it-yourself hardware/software.  Be aware, because there is no way to avoid this situation except to work with hardware and software that Digitrax approves.

In general, WRRwA code should probably allocate 128 bytes per message, and _guarantee_ that no incoming message "overflows" the buffer, whatever "guarantees" means.  Such a library change _could_ make a newer library version "not compatible" with certain hardware/software platforms if the new library is not compatible with the old code's memory usage and/or "timings", though.

kiwi64ajs, 2024-04-19:
Hi cpm64,

Most of the definitions in the ln-opc.h file pre-dated me and were done in 2001 by Ron W. Auld and they’ve not been much of a problem before now…

One way to provide the desired behaviour you’re looking for is to adopt a similar approach to the DCC-EX project, which uses this special pre-processor:

	#if __has_include ( "myAutomation.h”)

Optionally include a custom myAutomation.h file if it exists at build time.

If an appropriate file exists in the sketch folder, we may be able to do a similar thing to allow some behaviour to be changed at the Arduino sketch level.

It's probably worth giving it a try, as it would be useful for some other config-type items that alter the library's behaviour.

Regards

Alex Shepherd

m: +64-21-777764
e: ***@***.***