mrrwa/LocoNet #46: Uhlenbrock LNCV packets

fulda1, 2024-06-16:
Hi, this is not real issue,

Me and Paco F. Canada just spend some time playing with Uhlenbrock packets well known as LNCV. In reality used for lot of occasions, for example additional data about locomotive profile. You can extend your tables in:
https://github.com/mrrwa/LocoNet/blob/master/LocoNet.cpp#L1774

Here is my observation:
both ED 0F ... and E5 0F ... using same pattern.
Better to say, two patterns exist:

LNCV packet request (16 bit):
ED 0F [SRC] [DSTL] [DSTH] [REQID] [PXCT1] [CLSL] [CLSH] [LNCVL] [LNCVH] [VALL] [VALH] [FLAG]
LNCV packet response:
E5 0F [SRC] [DSTL] [DSTH] [REQID] [PXCT1] [CLSL] [CLSH] [LNCVL] [LNCVH] [VALL] [VALH] [FLAG]

DATA packet request (8 bit):
ED 0F [SRC] [DSTL] [DSTH] [REQID] [PXCT1] [D1] [D2] [D3] [D4] [D5] [D6] [D7]
DATA packet response:
E5 0F [SRC] [DSTL] [DSTH] [REQID] [PXCT1] [D1] [D2] [D3] [D4] [D5] [D6] [D7]

How to read:
1) PXCT1 represent 7th bits of D1 to D7 (or corresponding bytes for LNCV):
[PXCT1] = [0, D7.7, D6.7, D5.7, D4.7, D3.7, D2.7, D1.7]
they add them immediate and then manage as usual 8bit bytes.
2) REQID represent command that has to be done with data (I do not have full table)

[SRC] represent who is asking, I see following table:
00 = MASTER (command station - sometimes they use SPU)
01 = KPU (throttle - they use name KPU for hand throttle and build in one)
02 = no realized yet
03 = TWINBOX FRED
04 = IBSWITCH
05 = MODULE (any slave module, like occupancy detector)
..
08 = PC

[DSTH] & [DSTL] destination. Surprise, when they use ASCII, they have order LH, if use number, they have endianity HL.
InteliBox SPU = "IB" (DSTL="I", DSTL="B") (Daisy command station is using it as well)
InteliBox KPU = "IK" (Daisy throttle)
TwinBox = "xT" (DSTL=x can be any number 0 to 15; DSTH="T")
InteliSwitch KPU = "IS"
Module = 0005

[REQID]
0x03 = ?? (from your mail)
0x05 = ?? (from your mail)
0x06 = version request IB2 (no data)
0x07 = serial number request TwinCenter (no data)
0X08 = version response IB2 ([CLASS] contain version, [D3] to [D5]?)
0X09 = serial number response TwinCenter ([D1] to [D5] ?)
0x0B = Loco data response [CLASS] contain loco number, rest do not know
0x0D = Loco data read [CLASS] contain loco number
0x1F = CFG read (LNCV)
0x20 = CFG write (LNCV)
0x21 = CFG request (LNCV)

[FLAG]
0x01 = RO
0x40 = PROFF
0x80 = PRON
0xFE = LED control request
0xFF = Key change response

[CLASS]
it is 16 bit value from [CLSH]:[CLSL] - I guess they always understand D1 and D2 as 16 bit, but not sure, some materials present them as separate.

[LNCV]
again 16 bit of [LNCVH]:[LNCVL]

[VALUE]
again 16 bit of [VALH]:[VALL]

Some examples:
[ED 0F 01 49 42 0D 00 05 00 00 00 00 00 00 1F] ask loco 5
[E5 0F 00 49 4B 0B 04 05 00 52 11 05 00 00 5B] Loco 5 response (no details about meaning, probably icons on display etc)

[ED 0F 01 49 42 06 00 00 00 00 00 00 00 00 11]  ask for software version
[E5 0F 00 49 4B 08 00 26 10 01 01 01 00 00 28]  v1.026 ---> in the LCD when you look for software version shows v1.025 - v1.026 

[ED 0F 01 49 42 07 00 00 00 00 00 00 00 00 10] ask for serial number
[E5 0F 00 49 4B 09 00 20 00 00 59 31 00 00 56] serial number -> 2000005931

[ED 0F 01 49 42 05 00 0C 00 00 00 00 00 00 1E]   ???
[E5 0F 00 49 4B 03 00 0C 00 01 02 00 00 00 1B]   ???

fulda1, 2024-06-16:
Ach, forget to point to one nice doc: that describe few variants 
https://uhlenbrock.de/de_DE/service/download/handbook/en/I4B050BD-02F.apd/Bes69000-TechDesc.pdf

kiwi64ajs, 2024-06-16:
Hi Jindra,

Others added the LNCV protocol support, but I’ve never actually used it. I also don’t have compatible hardware to test any changes, nor have I tried to understand the mechanism in any detail.

So, if you believe it can be improved and are able to test the improvements, then feel free to submit the improvements via a PR or just email me the modified files, and I can commit them to the repo.

Regards

Alex Shepherd