mrrwa/LocoNet #48: Avr/io.h error when compiling on Arduino R4

HuibMaaskant, 2024-10-12:
Hi Alex,

My LocoNet project is compiling fine on an Arduino Uno R3, but when I compile the same code on an Arduino Uno R4, I get the following library error:

In file included from /Users/huib/Documents/Arduino/libraries/LocoNet/LocoNet.cpp:71:0:
/Users/huib/Documents/Arduino/libraries/LocoNet/utility/ln_sw_uart.h:47:12: fatal error: avr/io.h: No such file or directory
 #  include <avr/io.h>
            ^~~~~~~~~~
compilation terminated.
exit status 1

Compilation error: exit status 1

Any thoughts on how to get around this?

Thank you for your time!

Huib

kiwi64ajs, 2024-10-12:
Hi Huib,

> On 13 Oct 2024, at 4:09 AM, HuibMaaskant ***@***.***> wrote:
> Hi Alex,
> My LocoNet project is compiling fine on an Arduino Uno R3, but when I compile the same code on an Arduino Uno R4, I get the following library error:
> 
> In file included from /Users/huib/Documents/Arduino/libraries/LocoNet/LocoNet.cpp:71:0:
> /Users/huib/Documents/Arduino/libraries/LocoNet/utility/ln_sw_uart.h:47:12: fatal error: avr/io.h: No such file or directory
> 
> include <avr/io.h>
> 
>         ^~~~~~~~~~
> compilation terminated.
> exit status 1
> 
> Compilation error: exit status 1
> 
> Any thoughts on how to get around this?
> 
The short answer is to stick to the Uno R3 for now… ;)

The longer answer is that the Uno R4 has a very different CPU (Renesas) than the Uno R3 (AVR), and the LocoNet library is _VERY_ AVR specific. I don’t want to hack Renesas CPU support in that ugly old code base.

My preferred path is to migrate to using the LocoNet2 library, which I’ve been chipping away at over the years. It uses the hardware UART capabilities of the various CPUs to do the LocoNet comms and different techniques to detect LocoNet Traffic.

I started Uno R4 support recently, but it's very preliminary and still in development and hasn’t been used by others.

Also, things have been fixed in the LocoNet library that has not been ported over to the LocoNet2 library, but depending on what you want to do, it may work fine.

If you’re curious can download the library here: https://github.com/mrrwa/LocoNet2/archive/refs/heads/master.zip

There is an example that runs on the Uno R4 here: https://github.com/mrrwa/LocoNet2/blob/development/examples/Basic-Listener-UnoR4/Basic-Listener-UnoR4.ino , but be aware the Pins used for LocoNet will be different as its using the hardware UART now and that’s probably only available on specific pins.

It's not code compatible with the LocoNet library, as in swap the library and recompile, as the call-backs have to be explicitly registered, but once you see the example above it should give you some hints as to how to copy pase your existing code into the new structure.

Feel free to email me your sketch, and I can probably look at porting it over for you, as I need real-world examples to test what I’ve done as just Listening for LocoNet packets isn't much of a test… ;) 

HTH

Regards

Alex Shepherd

m: +64-21-777764
e: ***@***.***

HuibMaaskant, 2024-10-13:
Hi Alex,

Thank you for your elaborate answer. To cut a long story short… you’re saying forget about it ;-)

My project is about creating a RGBW LEDstrip controller and the PCB is designed to use specific pins, so that would mean creating a new PCB. So I’ll stick with the R3 for now.

Thanks a lot for the example. I’ll study it for future projects.

Grtz! Huib